dadamachines/doppler #1: Compile error for Arduino doppler example

pdp7, 2019-03-17:
Compilation error in the Arduino doppler example due to include for `/Users/svenbraun/Documents/GitHub/icicle/top.bin.h` which seems hardcoded for the original developers computer:
```
Arduino: 1.8.5 (Mac OS X), Board: "dadamachines doppler, Enabled"

/Users/helen/Library/Arduino15/packages/dadamachines - M4/hardware/samd/1.3.0/libraries/doppler-software/doppler/doppler.ino:2:62: fatal error: /Users/svenbraun/Documents/GitHub/icicle/top.bin.h: No such file or directory
 #include "/Users/svenbraun/Documents/GitHub/icicle/top.bin.h"
                                                              ^
compilation terminated.
exit status 1
Error compiling for board dadamachines doppler.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

Photos:
https://photos.app.goo.gl/2ngKTJ1svTuC6FXn9

pdp7, 2019-03-17:
This is the line that causes the error:
https://github.com/dadamachines/doppler-software/blob/a2925fce789746b4d4b8c13816c1a09f9b05f76b/doppler/doppler.ino#L2

pdp7, 2019-03-17:
Note: the other example that appears in the Arduino menu for the doppler library does compile OK:
`doppler-software/examples/hex-leds-test/hex-leds-test.ino`
https://github.com/dadamachines/doppler-software/blob/a2925fce789746b4d4b8c13816c1a09f9b05f76b/examples/hex-leds-test/hex-leds-test.ino

pdp7, 2019-03-17:
@nevvkid @noscene should I be creating issues like this issue in this repo or in https://github.com/dadamachines/doppler-software ? thanks!

nevvkid, 2019-03-17:
@pdp7 I think this repo should be the main to improve the doppler project. So all good. Keep the issues coming here ;) I hope we can fix all of them asap. 
Thanks for getting your hands "dirty" ;)