dadamachines/doppler #5: Froze during program upload and now the board is dead

mcclure, 2019-12-23:
Yesterday I took my Doppler out of the package, I wrote a very small program and uploaded it to the Doppler several times. It worked fine.

Today I added ICEClass.h and ice40.upload to the program to control the LEDs. Including ICEClass.h made the program much larger. It went from "5% used" to "20% used" or so. After adding ICEClass.h, uploads took much longer, and about half the uploads would halt partway through with an error message. I tried to upload, got an error, uploaded again, succeeded and the LEDs changed as desired. After making a change I made a third attempt to upload. **This time the upload halted partway through with an error. The Doppler never came back up again.** I have tried unplugging and replugging it from the mac and rebooting the mac. Its current behavior is:

- Nothing happens when I plug it in— no lights or anything (the last program uploaded would have cycled the lights)
- Device does not appear in tools->port menu in the Arduino editor
- If I press the button in the center of the doppler, the light near the USB cable flashes exactly once, but nothing else changes

What do I do?

I was using Arduino 1.8.5 when I started but upgraded to Arduino 1.8.10 while trying to debug. I am on OS X 10.13.6.

nevvkid, 2019-12-23:
Hi, @mcclure , 
did you try getting into the bootloader with double-pressing the reset? 
All SAMD51 boards still have problems with the upload on OSX Version of the Arduino IDE. 
I recently moved away from the IDE and use arduino-cli on the command line.
Which doesn't have issues with upload.

mcclure, 2019-12-23:
@nevvkid , thank you! double-pressing the reset fixed it.

I did not find a document explaining double-pressing the reset. Which document explains this? So I can consult it in future.

---

Also FYI : I tried "tools->burn bootloader" in Arduino to see what would happen and I got this error

```
Arduino: 1.8.10 (Mac OS X), Board: "dadamachines doppler, Enabled"

java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:329)
	at processing.app.Editor.lambda$handleBurnBootloader$65(Editor.java:2394)
	at java.lang.Thread.run(Thread.java:748)
Error while burning bootloader.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

closed 2019-12-23 by mcclure

nevvkid, 2019-12-23:
@mcclure 
Glad this worked. I was sure we wrote about the double-pressing the reset somewhere on the quickstart guide. But maybe it's missing. :( 
I gave some doppler boards to people who wanted to do more documentation and tutorials but sadly no results. 
Keep posting here or on the forum if you run into problems.  
Burning the bootloader requires a programming adapter. Like the Atmel ICE SAM.

mcclure, 2019-12-28:
Thank you, I have some further questions (including one mysterious compile error on something that should work). I posted here: https://forum.dadamachines.com/t/pin-headers-what-do-i-connect-the-doppler-pins-to/352/4?u=mcc